makerbase-mks/MKS-SKIPR #2: BTT TMC5160 v1.2 don't work via spi

sanek88lbl, 2022-08-23:
This drivers work in BTT SKR2 and FLY Gemini V1.
I have tried different speeds for SPI clock: from 100000 to 4000000 Hz.
Log from Fluidd Console:
```
12:19:56 // Klipper state: Ready
12:24:39 $ DUMP_TMC STEPPER=stepper_x
12:24:39 // ========== Write-only registers ==========
12:24:39 // GLOBALSCALER: 0000007d globalscaler=125
12:24:39 // IHOLD_IRUN: 00061f0a ihold=10 irun=31 iholddelay=6
12:24:39 // COOLCONF: 00000000
12:24:39 // PWMCONF: c40c001e pwm_ofs=30 pwm_autoscale=1 pwm_autograd=1 pwm_reg=4 pwm_lim=12
12:24:39 // TPOWERDOWN: 0000000a tpowerdown=10
12:24:39 // ========== Queried registers ==========
12:24:39 // GCONF: 00000000
12:24:39 // CHOPCONF: 00010000 tbl=2 mres=0(256usteps)
12:24:39 // GSTAT: 00000800
12:24:39 // DRV_STATUS: ff7ff000 s2vsa=1 s2vsb=1 stealth=1 fsactive=1 csactual=127 stallguard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
12:24:39 // FACTORY_CONF: 00004000
12:24:39 // IOIN: e00801ff refl_step=1 refr_dir=1 encb_dcen_cfg4=1 enca_dcin_cfg5=1 drv_enn=1 enc_n_dco_cfg6=1 sd_mode=1 swcomp_in=1 version=0xe0
12:24:39 // LOST_STEPS: 00000000
12:24:39 // MSCNT: f0040000
12:24:39 // MSCURACT: ffbfdfe0 cur_a=-32 cur_b=-65
12:24:39 // OTP_READ: 7fe00001 otp_fclktrim=1
12:24:39 // PWM_SCALE: 00400000 pwm_scale_auto=64
12:24:39 // PWM_AUTO: 0007fe00 pwm_grad_auto=7
12:24:39 // TSTEP: 00000000
12:24:49 $ INIT_TMC STEPPER=stepper_x
12:24:50 !! Unable to write tmc spi 'stepper_x' register GLOBALSCALER
``
[klippy.log](https://github.com/makerbase-mks/MKS-SKIPR/files/9401359/klippy.log)
`

sanek88lbl, 2022-08-24:
I found a mistake. Need to fix for all slots:
[image]
Or do it like in FLY Gemini:
[image]

gaby64, 2023-01-20:
Did you manage to fix it?

sanek88lbl, 2023-01-20:
> Did you manage to fix it?

Yes. I removed the SLP pin from the driver.

sanek88lbl, 2023-01-20:
[image]