makerbase-mks/MKS-SKIPR #6: CAN functionality

NPCarl, 2022-09-06:
Trying to query for canbus_uuid, but cannot because missing klippy_env. How can we make this work? 

`root@mkspi: # /klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0
-bash: /root/klippy-env/bin/python: No such file or directory`

`root@mkspi: # /klippy-env
-bash: /root/klippy-env: No such file or directory`

MKS-hosiey, 2022-09-14:
Is your canbus connected to the usb port of the motherboard? The can interface on the motherboard is not supported, you need a usb to can device, connect it to the motherboard, and then switch to can

NPCarl, 2022-10-17:
I am still unable to find canbus_uuid with a u2c device. i have tried both host and board usb.  any suggestions?

opiswahn, 2022-11-09:
@MKS-hosiey so you are saying the USB cable needs to be connected all the time? CAN is not enough?

G4Cab, 2022-12-03:
> Is your canbus connected to the usb port of the motherboard? The can interface on the motherboard is not supported, you need a usb to can device, connect it to the motherboard, and then switch to can

Why is there a CAN BUS connector and interface if it isn't supported???
How do we use the CAN interface to use for example a BTT EBB36?

G4Cab, 2022-12-06:
> her answer by mail https://github.com/makerbase-mks/MKS-THR36-THR42-UTC
> 
> please check it

and how is this helping with a SKIPR Board?

G4Cab, 2022-12-21:
https://klipper.discourse.group/t/mks-skipr-can-bus/5377

Radulfus, 2023-01-06:
If you follow G4Cabs link to klipper.discourse you'll see our progress in the issue analysis.
I have measured the CAN-bus with an oszilloskop and a logic analyzer.
We have no CAN-bus traffic, no signals from the MCU on pins PB12/13 to the transceiver.
So, IMHO the MCU (STM32F407) is not proper initalized for CAN-bus communication. The BTT Octopus Board has a similar CAN-bus hardware implementation and CAN-bus works flawless. The only difference between MKS SKIPR and BTT Octopus is the MCU Type (STM32F446).
I suppose somthing between this two MCU is differnt related to the CAN-bus configuration. That could be the fault.
So IMHO it is up to the Klipper developers to help.

Radulfus, 2023-01-15:
Issue is solved!
https://klipper.discourse.group/t/mks-skipr-can-bus/5377/90?u=cosmicos