makerbase-mks/MKS-SKIPR #7: Neo pixel stealth burner lights not working

Shamoon78, 2022-10-06:
I swapped the octopus and raspberry pi on my voron 2.4 to MKS SKIPR everything working fine except the neo pixel leds on stealth burner (3 leds ) i connected the 5 volt, ground and and PC5 on Npixel connector but no luck . any help appreciated.

Shamoon78, 2022-10-06:
Update connecting the middle wire of neopixel to exp1_6 (PD8) instead of PC5 and configure accordingly in printer.cfg the leds are working fine but why using PC5 pin is not working ??!! please let me know . thanks because i need the both expansion connectors for the screen .

MKS-hosiey, 2022-10-12:
pc5 is the s pin of the neopixel, as long as the configuration is correct, the wiring is correct, the pin is not burned out, it can work

Shamoon78, 2022-10-12:
i dont know if it is burned or not but i try every pin is working except pc5 not working now iam using the can pin ,using pt1000 pin also work .

Shamoon78, 2022-10-12:
the wiring 100% correct because all other pins work and can change the color easy .except pc5

nehilo, 2022-10-22:
I have the same problem. Not working by PC5 pin

gaby64, 2023-01-07:
I have the same problem aswell, neopixel doesn't work on the neopixel port.

brutzler, 2023-01-08:
same here. PC5 is only working randomly.

And to say: why is this plug different to the others? 
Neopixel (J26): 5V-SIG-GND 
All others in the same area (J22...J25): 5V-GND-SIG

Can I take the 5V for 3 Neopixels directly from J26?

brutzler, 2023-01-10:
> pc5 is the s pin of the neopixel, as long as the configuration is correct, the wiring is correct, the pin is not burned out, it can work
I doubt this statement.
I tested the same as shamoon78. PD8 and PD11 (only tested these) are working proper. PC5 is NOT working. Imho the electrical design (resistors/diodes) around the port is disturbing.

gaby64, 2023-01-20:
For new boards manufactured in 2023 this is supposed to be solved.
For everyone else, MKS has provided a fix, I have tested it and confirmed it works:

[image]
[image]
[image]

Printhing3d, 2023-03-13:
I had the same problem. PC5 pin is not working. I use BLtouch pin, PA8 , working like a charm.