makerbase-mks/MKS-SKIPR #8: How to get Printer.cfg example for Cartesian Printer?

gabber130779, 2022-10-07:
Hello, 

is it possible to get a example printer.cfg for a Cartesian Printer. ther is only one for Voron 2.4

Shamoon78, 2022-10-08:
MKS made this nice board but there support to help peoples out fixing problems is absent .very sad.

MKS-hosiey, 2022-10-12:
> 

> MKS made this nice board but there support to help peoples out fixing problems is absent .very sad, if there support is helpfull, I am going to order in bulk to Australia. but no way.

If you encounter problems, you can contact the technicians online on facebook

Shamoon78, 2022-10-12:
facebook link?

Townie-au, 2022-11-19:
You can just edit the printer.cfg file.
I've got everything working on Ender 3 except stock LCD screen. 
I can't work out the pins

closed 2022-11-19 by gabber130779