makerbase-mks/MKS-SKIPR #14: Error "Internal error during connect: bad marshal data (tuple size out of range)"

danielbrownjr, 2022-12-02:
Encountered an error 
```Internal error during connect: bad marshal data (tuple size out of range)```

See log: [klippy.log](https://github.com/makerbase-mks/MKS-SKIPR/files/10141867/klippy.log)

G4Cab, 2022-12-04:
Did you flashed the latest Armbian to your sd/emmc ?

https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e?usp=sharing

rooh00, 2023-01-18:
Mesmo problema após atualizar, alguma solução?

mozg4D, 2023-01-28:
same problem

dhnsilva, 2023-03-17:
Same problem here!

mozg4D, 2023-07-06:
update everything except moonraker in settings (bottom of the page), restart firmware

h4natural, 2023-07-11:
I've solved!
First, reformat a sdcard/eMMC to FAT32 after clean all partitions.
And then flash the Armbian_klipper image file, use it to the SKIPR board

closed 2026-09-10 by danielbrownjr