makerbase-mks/MKS-SKIPR #18: Make CAN interface work properly (Not only CAN-to-USB !!!!)

opiswahn, 2022-12-09:
Currently, only CAN-to-USB communication is possible with the SKIPR board. This really sux, as the board has a CAN  connector which is not functional as it seems. 
I could get my FLY-SB2040 to work in CAN-to-USB mode, but for that I would not have needed the SKIPR board, this would have worked with any other. You are giving away the selling point for your board. PLEASE DO SOMETHING.

G4Cab, 2022-12-13:
+1

if i read the schematics correctly, the can bus is on the stm32.
so maybe we are never possible to use it with klipper the way we thought to use it.
klipper has to access the can bus via the stm32, and that's not possible at this moment as i know.

opiswahn, 2022-12-21:
yea, i saw that... So from what i understand it is NOT possible to use CAN ATM correct? I Still don't understand why an extra USB-To-CAN board would be necessary, the SKIPR should do this.

Nakajima-Kuro, 2022-12-26:
From what I see, BTT Octopus also have Canbus on STM32 and it work fine with SB2040 (Can H and L on the CAN interface and no need for extra USB cable between the toolhead and the board) so this is really weird how it does not work with MKS Skipr

G4Cab, 2023-01-15:
Just for Info, big process on discourse.
https://klipper.discourse.group/t/mks-skipr-can-bus/5377

we realized that mks used the second can interface on the skipr.
Kevin o'Connor had found a little problem on the stm32
First tests are promising, and i have a connection via the onboard can connector to my toolhead 😎

G4Cab, 2023-01-15:
Related issue: https://github.com/makerbase-mks/MKS-SKIPR/issues/12