makerbase-mks/MKS-SKIPR #19: Can't MCU connection dfu mode

pure100kim, 2022-12-18:
I connected usb to mcu type-c port. and then
pressed the boot0 switch and turned on the power button, but the dfu usb is not recognized at all.
It is not recognized as STM32CUBE. Not even ST-LINK.
It would be nice if it was a shunt type, but it's too inconvenient to hold down the switch.

MKS-hosiey, 2022-12-29:
Is your data cable connected? There are two type_c interfaces on the motherboard, you need to connect to the MCU interface

pure100kim, 2023-01-10:
> Is your data cable connected? There are two type_c interfaces on the motherboard, you need to connect to the MCU interface

Of course. I used data cable including your acc in the box. And also used cell phone data cable.