makerbase-mks/MKS-SKIPR #20: TMC2209 wiring for use UART and sensorless homing ?

Montana758, 2022-12-19:
Hello, I hope you can help me. 
Which pins from the TMC need to be connected to which pin on the board? 
I would like to use the drivers in uart mode.

G4Cab, 2022-12-21:
Use this pins for your jumper beneath the tmc2209
[image]

Montana758, 2022-12-21:
Thanks, I set this jumper.
But the sensorless homing doesn't work.
Do I have to connect a cable from the TMC2209 to one of the pins of "5" Sensorless Homing (on the picture) connector?

[image]

here my printer.cfg:

[stepper_x]
step_pin:PC14
dir_pin:!PC13
enable_pin:!PC15
microsteps: 16
rotation_distance: 40
full_steps_per_rotation:200  #set to 400 for 0.9 degree stepper
#endstop_pin:!PA14
endstop_pin: tmc2209_stepper_x:virtual_endstop
#position_min: 0  
position_endstop: 0
position_max: 230
homing_speed:50
homing_retract_dist:0
#homing_positive_dir:true
step_pulse_duration:0.000008

[tmc2209 stepper_x]
uart_pin: PE6
diag_pin: PA14
run_current: 0.3
hold_current: 0.5
interpolate: True
driver_SGTHRS: 10
stealthchop_threshold: 1

gaby64, 2023-01-06:
If im not mistaken, you need a jumper on the Diags header, that will bridge the sensorless homing of an axis to the jumpered endstop.

Icesythe7, 2023-02-18:
Anyone get this working?

gaby64, 2023-02-18:
> Anyone get this working?

Yes

Icesythe7, 2023-02-18:
I set the jumper on the board under the driver, set the jumper on diag, set it in config still wont trigger :(
`[stepper_x]
step_pin:PC14
dir_pin:!PC13
enable_pin:!PC15
microsteps: 64
rotation_distance: 40
full_steps_per_rotation: 200
endstop_pin: tmc2209_stepper_x:virtual_endstop
homing_retract_dist: 0
position_endstop: 120
position_max: 120
homing_speed: 40` 

`[tmc2209 stepper_x]
uart_pin: PE6
diag_pin: ^PA14
run_current: 0.55
sense_resistor: 0.110
stealthchop_threshold: 0
interpolate: false
driver_SGTHRS: 90 # 255 is most sensitive value, 0 is least sensitive`

nema 17's using ez2209 drivers from btt

gaby64, 2023-02-18:
[image]

Assuming you put a jumper on that diags header in the lower right. First, I would remove the pullup (^) on the diag pin, `diag_pin: PA14`. Second I would increase the `driver_SGTHRS` value to 130 until you hage a successful trigger.

Icesythe7, 2023-02-18:
The motors won't even move now, removing the pull up resistor as u said just make it auto trigger endstop, inverting diag with ! also makes it auto trigger engstop, currently if I home nothing happens yet it says i homed and i can move as per normal with the arrows yet I never actually homed so position is wrong

gaby64, 2023-02-18:
Lower driver_SGTHRS until it does not trigger

gaby64, 2023-02-18:
https://www.klipper3d.org/TMC_Drivers.html#find-highest-sensitivity-that-successfully-homes

zu3st-de, 2023-03-17:
@gaby64:
Where did you get the pinout document? 
I try to find some documents like this but without succsess.

gaby64, 2023-03-17:
> @gaby64: Where did you get the pinout document? I try to find some documents like this but without succsess.

https://github.com/makerbase-mks/MKS-SKIPR/blob/main/hardware/MKS%20SKIPR%20V1.0_002/MKS%20SKIPR%20V1.0_002%20PIN.pdf

eldeeb91, 2023-04-17:
I have sensorless homing working on 4 steppers. Here is my config:
```
[stepper_x]
step_pin: STEP_1
dir_pin: !DIR_1
enable_pin: !EN_1
endstop_pin: tmc2209_stepper_x:virtual_endstop
rotation_distance: 40
microsteps: 64
full_steps_per_rotation: 200
#gear_ratio:
step_pulse_duration: 0.000000100
position_min: -2.5
position_endstop: -2.5
position_max: 600
homing_speed: 50
homing_retract_dist: 0
#homing_retract_speed:
#second_homing_speed:
#homing_positive_dir:

[tmc2209  stepper_x]
uart_pin: UART_1
diag_pin : ^EndStop_1
interpolate = True
run_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 500
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 20
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 0
driver_HSTRT: 5
driver_PWM_AUTOGRAD: True
driver_PWM_AUTOSCALE: True
driver_PWM_LIM: 12
driver_PWM_REG: 8
driver_PWM_FREQ: 1
driver_PWM_GRAD: 14
driver_PWM_OFS: 36
driver_SGTHRS: 70

[stepper_y]
step_pin: STEP_2
dir_pin: DIR_2
enable_pin: !EN_2
endstop_pin: tmc2209_stepper_y:virtual_endstop
rotation_distance: 40
microsteps: 16
full_steps_per_rotation: 200
#gear_ratio:
step_pulse_duration: 0.000000100
position_min: -3.5
position_endstop: -3.5
position_max: 515
homing_speed: 25
homing_retract_dist: 0
#homing_retract_speed:
#second_homing_speed:
#homing_positive_dir:

[tmc2209  stepper_y]
uart_pin: UART_2
diag_pin: ^EndStop_2
interpolate = True
run_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 500
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 20
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 0
driver_HSTRT: 5
driver_PWM_AUTOGRAD: True
driver_PWM_AUTOSCALE: True
driver_PWM_LIM: 12
driver_PWM_REG: 8
driver_PWM_FREQ: 1
driver_PWM_GRAD: 14
driver_PWM_OFS: 36
driver_SGTHRS: 45

[stepper_y1]
step_pin: STEP_3
dir_pin: !DIR_3
enable_pin: !EN_3
endstop_pin: tmc2209_stepper_y1:virtual_endstop
rotation_distance: 40
microsteps: 16
full_steps_per_rotation: 200
#gear_ratio:
step_pulse_duration: 0.000000100
# position_min: 0
# position_endstop: 0
# position_max: 550
# homing_speed: 50
# homing_retract_dist: 0
#homing_retract_speed:
#second_homing_speed:
#homing_positive_dir:

[tmc2209  stepper_y1]
uart_pin: UART_3
diag_pin: ^EndStop_3
interpolate = True
run_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 500
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 20
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 0
driver_HSTRT: 5
driver_PWM_AUTOGRAD: True
driver_PWM_AUTOSCALE: True
driver_PWM_LIM: 12
driver_PWM_REG: 8
driver_PWM_FREQ: 1
driver_PWM_GRAD: 14
driver_PWM_OFS: 36
driver_SGTHRS: 45

[stepper_z]
step_pin: STEP_4
dir_pin: DIR_4
enable_pin: !EN_4
#endstop_pin: tmc2209_stepper_z:virtual_endstop
endstop_pin:!EXP1_7
rotation_distance: 8
microsteps: 64
full_steps_per_rotation: 200
#gear_ratio:
step_pulse_duration: 0.000000100
position_min: 0
position_endstop: 0
position_max: 80
homing_speed: 1
homing_retract_dist: 0
#homing_retract_speed:
#second_homing_speed:
#homing_positive_dir:

[tmc2209  stepper_z]
uart_pin: UART_4
diag_pin: EndStop_4
interpolate = True
run_current: 0.8
sense_resistor: 0.110
stealthchop_threshold: 500
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 20
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 0
driver_HSTRT: 5
driver_PWM_AUTOGRAD: True
driver_PWM_AUTOSCALE: True
driver_PWM_LIM: 12
driver_PWM_REG: 8
driver_PWM_FREQ: 1
driver_PWM_GRAD: 14
driver_PWM_OFS: 36
driver_SGTHRS: 35

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB2,  EXP1_3=PE11, EXP1_5=PD9, EXP1_7=PE15, EXP1_9=<GND>,
    EXP1_2=PE10, EXP1_4=PD10, EXP1_6=PD8, EXP1_8=PE7,  EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PA6, EXP2_3=PE9, EXP2_5=PE8, EXP2_7=PD13,  EXP2_9=<GND>,
    EXP2_2=PA5, EXP2_4=PA4, EXP2_6=PA7, EXP2_8=<RST>, EXP2_10=<3.3v>,
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp1"

    # See the sample-lcd.cfg file for definitions of common LCD displays.

    STEP_1=PC14, DIR_1=PC13, EN_1=PC15, EndStop_1=PA14, UART_1=PE6,
    STEP_2=PE5, DIR_2=PE4, EN_2=PD14, EndStop_2=PA15, UART_2=PE3,
    STEP_3=PE1, DIR_3=PE0, EN_3=PE2, EndStop_3=PB15, UART_3=PB7,
    STEP_4=PB5, DIR_4=PB4, EN_4=PB6, EndStop_4=PA13, UART_4=PB3,
    STEP_5=PD6, DIR_5=PD5, EN_5=PD7, EndStop_5=PC5, UART_5=PD4,
    STEP_6=PD2, DIR_6=PD1, EN_6=PD3, EndStop_6=PB14, UART_6=PD0,
    STEP_7=PC7, DIR_7=PC6, EN_7=PC8, UART_7=PD15,
```

prondz, 2023-04-19:
Thank you so much, reading your code I was able to find out where I was going wrong. My printer is live again

mruviaro, 2023-06-09:
Could someone send the sensorless configuration? did you have to put any jumpers on the board?

gaby64, 2023-06-12:
read the previous posts!

mruviaro, 2023-06-12:
> leia os comentários anteriores! está tudo aí

From what I understand the configuration would be... ...but don't works...

[stepper_y]
step_pin: PE5
dir_pin: !PE4
enable_pin: !PD14
microsteps: 16
rotation_distance: 40
full_steps_per_rotation: 200  #set to 400 for 0.9 degree stepper
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_endstop: 0
position_max: 310
homing_speed: 50
homing_retract_dist: 0
homing_positive_dir: false
step_pulse_duration: 0.000008

[tmc2209 stepper_y]
uart_pin: PE3
diag_pin: ^PA15
driver_SGTHRS: 0
run_current: 0.8
hold_current: 0.5
interpolate: True

gaby64, 2023-06-12:
Did you set the jumpers as indicated in other posts?

What do you mean by it does not work? What does it do?

driver_SGTHRS: 0
This value would never trigger, it will crash.

mruviaro, 2023-06-12:
sorry, i'm used this (sgthrs 0 and 255) config before to test, but now is 130....

moswald14, 2023-09-16:
Thanks so much. never considered X- and Y- were the pins I was trying to find

BETLOG, 2024-04-11:
[Same problem](https://klipper.discourse.group/t/uart-sensorlesss-homing-assistance-please/15661). 
My setup looks like it should work, but doesn't. Extremely frustrating.
I guess I need to start switching all my 2209s and 2226s into X socket to test if any of them work.