makerbase-mks/MKS-SKIPR #21: MKS-SKIPR sd card printing

DapDeveloper, 2022-12-20:
Hi,
I installed the new board mks-skipr on my printer.
I need to print using the sd card like a normal 3d printer.
I tried to insert the card on the TF-HOST and TF-MCU but i don't see
the gcodes from the Fluid interface or tft35.
How can i configure the board to read SD card ?
Thanks

G4Cab, 2022-12-21:
create automount rules in your armbian environment, this way you should be able to use the host-sdcard-slot

ant0nyk1ng, 2023-02-22:
> create automount rules in your armbian environment, this way you should be able to use the host-sdcard-slot

G4Cab ... where can I find a step by step instructions to do this? I tried googling but I couldn't find anything that made sense to me.

ARTYUSS, 2023-06-11:
HI, 
I would be interested to, it there anysolution to use the sd slot for direkt printing ?

Futtawuh, 2024-11-10:
> HI, I would be interested to, it there anysolution to use the sd slot for direkt printing ?

@ARTYUSS did you figure this out?