makerbase-mks/MKS-SKIPR #31: mcu shutdown overload

chriskleov, 2023-02-07:
Durinig printing i receiving "MCU 'mcu' shutdown. Timer too close This often indicates the host computer is overloaded". printer during stantby mem 87%. camera is connected it and no work. use ethernet net connection

Khager63, 2023-02-11:
Same issue here. Printer stops afret 30min of printing. Ethernet connection, 16G Emmc.TMC2209 drivers.

MCU 'mcu' shutdown: Timer too close
clocksync state: mcu_freq=168000000 last_clock=3155556439005 clock_est=(18768.342 3150321636225 168003796.706) min_half_rtt=0.000252 min_rtt_time=18539.490 time_avg=18768.341(945.603) clock_avg=3150321636225.860(158864888757.643) pred_variance=184085513.786
Dumping serial stats: bytes_write=18560422 bytes_read=6546157 bytes_retransmit=615 bytes_invalid=0 send_seq=462021 receive_seq=462018 retransmit_seq=462020 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=325 stalled_bytes=0
[klippy.log](https://github.com/makerbase-mks/MKS-SKIPR/files/10712633/klippy.log)

chriskleov, 2023-02-26:
Did you use mainsail?

Khager63, 2023-02-26:
I used Fluidd. 
No errors anymore, flashed my MCU stm32f407 with this instuction.
When running "make menuconfig", select the 48KiB
# bootloader, and enable "Serial for communication" and select the "on USART1 PA10/PA9"
Now its OK, no errors anymore.