makerbase-mks/MKS-SKIPR #33: Cant reach the board over WiFi

PinGuiNPL, 2023-02-15:
Hi, 
I've been trying to solve a problem since yesterday. 
The connection works perfectly with a LAN cable, but the attempt to connect to the WIFI fails. 
Only if I enter the command "sudo dhclient -v" via (LAN)SSH does the board find the IP that I have already set in the router for the WiFi connection. 
Unfortunately, even then I can't connect properly, when I enter the IP, the Mainsail interface shows the window "Connecting to 192.***.*.**", alternating with "initializing" but unfortunately it never works beyond this window.
I can then also connect via WiFi through SSH, but unfortunately with errors, as if I had no access to the storage space. Can't make updates aso.

What ive tried:
- configured "wpa_supplicant-wlan0.conf"
- added "timeout 5" to the "dhcpcd.conf"
- edited "/etc/network/interfaces"
- edited "/etc/netplan/50-cloud-init.yaml"
- edited "/etc/network/interfaces.d/wlan0"
- installed realtek drivers

I don't know what to try next. I don't want to leave the printer connected to the LAN cable because I would have to leave a 10m long cable halfway through the room.
I'm using the "comfast cf-wu810n" WiFi Stick that comes with the board.

mozg4D, 2023-02-16:
will probably not help, but you must set not empty password for wifi to use "wpa-psk" in your router settings (192.168.1.1)

PinGuiNPL, 2023-02-17:
> will probably not help, but you must set not empty password for wifi in your router settings (192.168.1.1) to use "wpa-psk"

I've always had it set that way.

BUT i found a solution...
Armbian_23.02.0-trunk_Mkspi_bullseye_current_5.15.91.img

It's a community image which is much more up-to-date than that of MKS. This, with a small modification to make my 7 inch display work, eliminated all other problems I had with the official image.

Here the github link for this project:
https://github.com/redrathnure/armbian-mkspi

closed 2023-02-17 by PinGuiNPL