makerbase-mks/MKS-SKIPR #36: 'MCU': Unable to connect after power cycle

PinGuiNPL, 2023-02-28:
Hi, until this morning I was more than satisfied with the board. Before that I had a SKR 1.4 Turbo board with a Pi4B. In the last few days I have converted everything to the SKIPR and have already started calibrating the printer, as I have also made a couple of other upgrades. Everything was going great until yesterday, then I switched off the printer and wanted to continue working on the calibration today.... but unfortunately the printer no longer works. After switching on, Klipper no longer sees the MCU.:
```
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
```
Nothing changed in the config during the night.
Nevertheless, I checked these and the path is still: 
```
[mcu]
serial: /dev/ttyS0
restart_method: command
```
My first try was a different SD card for the MCU, the second with the mks_skipr.bin from MKS Github and the third with my own bin with make menuconfig... None of that worked.
To me it looks like some defect. If so, that was my first and last MKS board, I have 3 other printers with SKRs and have never had a broken board. But maybe someone here has an idea I didn't think of.

kino13, 2023-07-11:
Hello, 

I got exactly the same error since yesterday, did anyone find anything else or has mks answered?

Townie-au, 2023-08-03:
Hi
I also have the same issue

Townie-au, 2023-08-03:
I doubt MKS will respond.
It's usually the community that works the solution.

zadnezz32, 2023-09-12:
I'm in the same situation. No LEDs light up on the MCU side of the car. and the serial doesn't work.

misssiek, 2023-09-21:
my klipper have the same error, after shut down and power up, nothing can't help even reinstal.
[image]

kino13, 2023-09-24:
Hello all,

I know that many of you are having issues of different types, but
seriously, redo the cables of the canbus data from scratch with some good
cables (I used silicone wire 24AWG).
I have installed several can boards since I made this post, and I can
assure you 90% of the issues is always the cable or the physical connection
of the cable.
Another important thing, when the connectivity fails, klipper and even the
hosted linux will act in strange ways: losing the can0 interface, not
detecting the CAN board when the one failing is the secondary, hanging up,
I have seen it all and it was always the cables.

DK4YA, 2023-10-20:
> Hello all, I know that many of you are having issues of different types, but seriously, redo the cables of the canbus data from scratch with some good cables (I used silicone wire 24AWG). I have [...]

how helpful. All the people before you reported that the connection between host and MCU is not working. This is a direct connection on the board itself. What does that have to do with cable quality? Activating the CAN0 is still a long way off. I have to say that it was much better and easier on the BTT Manta board.

kino13, 2023-10-20:
Take it as you want, I have experienced all kinds of issues that got fixed
after replacing the cables on the canbus, and yes, I am referring to the
MCU.
For sure if the issues are happening even before reaching that point, you
will need to look elsewhere, but I saw all kinds: hang ups,
unresponsiveness, lost of MCU, lost of canbus (obviously), lost of usb
detection... And all of them were due to a bad canbus cabling.

kino13, 2023-12-07:
> My problem started the same as you, but it got worse, now only the LEDs
> turn on and the board stays frozen, and in serial mode when I connect to
> the assigned COM, it also stays frozen, does anyone have a solution, I
> think it is poor quality of the plate,
>
> The monster 8 is already in its third revision, same problems...

I moved to BTT boards on all my printers, I know that does not solve the
issue at hand, but I have not had any problem with them and their
documentation is excellent.

Lones01, 2024-02-01:
> my klipper have the same error, after shut down and power up, nothing can't help even reinstal. [...]

Hi
I also have the same issue

alexvnesta, 2024-07-01:
Does anyone have a solution to this issue?