makerbase-mks/MKS-SKIPR #37: HDMI doesn't work

yammaster, 2023-03-02:
What's necessary that HDMI works? If I connect the IPS50 display via HDMI interface, nothing will shown. Is there anything to configure?

PinGuiNPL, 2023-03-02:
SSH into the board and type
```
sudo nano /etc/X11/xorg.conf.d/40-monitor.conf
```
Put this there:
```
Section "Monitor" 
Identifier "HDMI-1"
Option "PreferredMode" "720x480"
EndSection
```
save.... then
```
sudo service KlipperScreen restart
```
Try different resolutions, till it works :)

yammaster, 2023-03-02:
... should I configure printer.cfg that it shows on HDMI?

EmJay276, 2023-03-30:
@yammaster did you manage to get it work?

yammaster, 2023-03-31:
Hello, not yet the hints doesn't solve the problem. But I've seen, there is a new image available where the HDMI is special mentioned.RegardsRene--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

EmJay276, 2023-03-31:
If I'll get it to work, I will let you know.

EmJay276, 2023-04-01:
@yammaster I figured it out! I had to flash the new image 
"Armbian-makerbase-k5.16.20-EN-20230301.img" onto the eMMC to get the IPS50 working!

closed 2024-02-01 by yammaster

EmJay276, 2024-02-01:
> Yes, I got the IPS working.

How?

yammaster, 2024-02-01:
I followed this guide, without the x-server calibration part and it works:

https://github.com/NexGen-3D-Printing/MKS-SKIPR-SETUP/blob/main/IPS50%20Fix