makerbase-mks/MKS-SKIPR #40: ADXL345 & PI-TS35 Problem

zu3st-de, 2023-03-26:
Hello,
if I connect the ADXL345 as described in the wiki, the PI-TS35 conducts white all the time.
Ist that normal?

DaTTcz, 2023-04-22:
I have the same problem in a combination of MKS PI and MKS MONSTER8v2. I was just thinking about buying a MKS SKIPR so now I know it has the same problem...

LeojoyMS, 2023-04-27:
Same in Skipr here

VORONENTHUSIAST, 2023-07-03:
Hey guys - did any of you solve this?

If not - I have! ;)

I was using 22awg wire for my SKIPR adxl345 connection - BAD IDEA.
I have since swapped out to a cut up ethernet cable - this has provided the screen whilst the adxl345 is plugged in, I no longer have the white screen of doom.

Nahkampfsocke, 2023-11-23:
@Voronenthusiast 
Hi  i  have the same Problem ,..   i also us a ethernet cable and  the  display turns white,..   pleas  tell mi your  exact wiring ( position an wire color