makerbase-mks/MKS-SKIPR #47: CAN BUS skipr

halk415, 2023-05-25:
Hello. I got into the hands of mks skipr. I am engaged in the manufacture of printers and I am looking for a good fee for myself. Mks skipr, good compact board, productive board. But I need it to work completely, help me set it up and I'm ready to buy them in large quantities!!! The essence of the problem: can bus does not work normally. It was possible to connect external boards via the can connector only if you connect the MCU to the HOST via an external USB cable, but this is not convenient and there are no free usb. How to make everything work normally? I did it according to your instructions on your website.

VORONENTHUSIAST, 2023-07-03:
It's been worked out - google it, you'll find the answer, just not from MKS

halk415, 2023-07-04:
> It's been worked out - google it, you'll find the answer, just not from MKS

I've already tried Googling, but I haven't found a solution to the problem. Can you drop the link?

VORONENTHUSIAST, 2023-07-04:
This was the top hit on google when searching "can bus MKS SKIPR":

https://klipper.discourse.group/t/mks-skipr-can-bus/5377/90

Have a read through all of it, they explain a lot of stuff. 

The following link is from KevinOConner (Klipper creator as far as I can work out): https://github.com/Klipper3d/klipper/pull/5988 You will find this mentioned in the original link. 

This is known to work with the Fysetc SB TH, and the Mellow SB2040
I have yet to implement it, but am ordering the parts this week - let me know how you get on!!

VORONENTHUSIAST, 2023-07-04:
Oh, thought I should add - as it seems it was Klipper related and not hardware related, that's probably why MKS haven't answered - from reading the thread the board is setup properly, it's just a case of recompiling firmware and such.

halk415, 2023-07-05:
> This was the top hit on google when searching "can bus MKS SKIPR": https://klipper.discourse.group/t/mks-skipr-can-bus/5377/90
> 
> Have a read through all of it, they explain a lot of stuff, the fact you didn't find it perplexes me - did you really look?! ;)
> 
> The following link is from KevinOConner (Klipper creator as far as I can work out): [Klipper3d/klipper#5988](https://github.com/Klipper3d/klipper/pull/5988) You will find this mentioned in the original link.
> 
> This is known to work with the Fysetc SB TH, and the Mellow SB2040 I have yet to implement it, but am ordering the parts this week - let me know how you get on!!

I found it, but nothing worked. 

I could only when I connected the USB. The STM controller is connected to the processor of the board on which the clipper is installed. In the board, the controller is connected via UART and I could not start it.
[image]

VORONENTHUSIAST, 2023-07-05:
Is your USB cable a data cable or just power? Have you updated to the latest Klipper? This was all merged around a month ago so is definitely working - I'm sorry to say, if yours isn't you are doing something wrong.

feros2011, 2023-08-29:
> Ваш USB-кабель предназначен для передачи данных или просто питания? Вы обновились до последней версии Klipper? Все это было объединено около месяца назад, поэтому оно определенно работает. К сожалению, если у вас это не так, вы делаете что-то не так.

Hi, friend! I have the same problem as the halk415. Without a USB cable, as connected by the author of the article, it does not work for me. Latest klipper firmware, configured the firmware for MCU and sb2040, but does not work directly, tell me how to configure the firmware for MCU and SB2040 correctly. I will be eternally grateful!!!

derTimm0, 2023-12-23:
i "belive" the connection via cunbus works but iam get no connection to the mcu part :-/ how i set the serial(?) conection to the skipr on printer.cfg. if i try to get the mcu id it will nothing show on USB/uart (via kiauh)

datarius2, 2024-08-14:
install:
`pip3 install pyserial`