makerbase-mks/MKS-SKIPR #50: Wifi

gallade785, 2023-05-29:
I can't connect to wifi. I use several adapters: COMFAST CF-WU815N; MERCUSYS MW150US; noname on MT7601U; noname on RTL8188FTV. And on skipr no one's don't work, but on my home pc with manjaro each work correctly. What should i do?

manuelchaves, 2023-06-29:
hello, here I use a prehistoric RALINK RT5370 of those Chinese very poor dongue. But the RALINT MT7601 also works, they don't weigh on the CPU

EmJay276, 2023-07-03:
Makerbase send me this overview of currently supported WiFi chipsets.
[image]

VORONENTHUSIAST, 2023-07-03:
I'm using COMFAST CFWU810N RTL8188EUS on 2 SKIPR boards, it worked straight out of the box.

From the screen shot above, I assume its parts of the rtl8188eu family.

qwertymanzero, 2023-08-17:
> hello, here I use a prehistoric RALINK RT5370 of those Chinese very poor dongue. But the RALINT MT7601 also works, they don't weigh on the CPU

How did you get MT7601 to work? I currently have that wifi chip and it will not show in armbian-config?
Device is connected in sudo lsusb, and forced to auto boot. no luck