makerbase-mks/MKS-SKIPR #59: unable to activate CAN=

DK4YA, 2023-09-23:


derTimm0, 2023-12-23:
i "belive" the connection via cunbus works but iam get no connection to the mcu part :-/ how i set the serial(?) conection to the skipr on printer.cfg. if i try to get the mcu id it will nothing show on USB/uart (via kiauh)

rt1970, 2024-01-13:
i have also no such device. My board is MKS THR42. If i try create Firmware it will create a .bin file.
But THR42 will .uf2 file. I have flash klipper.uf2 from THR firmware (can) by press "Boot" Button, then i plug USB in my PC and i copy the File to the new device with 128MB size. It closed the explorer. Now i dont have a serial port. That should be fine...
what klipper.bin config should i use on SKIPR?

rwetzel67, 2024-09-11:
Any answer on this? Same here: creating the file /etc/network/interfaces.d/can0 does nothing...
can0 is not available. Anything else that whould be done??