makerbase-mks/MKS-SKIPR #62: TMC2209 UART "Unable to read tmc uart 'stepper_x' register IFCNT"  ERROR

QTronics, 2023-10-25:
I am getting an "Unable to read tmc uart 'stepper_x' register IFCNT" 

My Config is 

[tmc2209 stepper_y]
uart_pin: PE3
run_current: 0.8
hold_current: 0.5
interpolate: True
stealthchop_threshold: 400

I have the UART Jumpers set in the correct position under the driver module.

I have tried all of the drivers in every slot and they all give me the same error.

iagobispo, 2023-11-16:
> I am getting an "Unable to read tmc uart 'stepper_x' register IFCNT"
>
> My Config is
>
> [tmc2209 stepper_y] uart_pin: PE3 run_current: 0.8 hold_current: 0.5 interpolate: True stealthchop_threshold: 400 [...]

managed to solve?

QTronics, 2023-11-16:
The issue was that I was running my hot end fans off the +24V with the ground going to the signal pins on the board for them, but the fan jumper on the board was set to 5V. I am not sure why that would make any difference, but I found that I was getting noise on the 5V supply, and it was causing this issue. Once I moved the Jumpers to 24V the issues were resolved.
 My Hot End is a common 24V setup, so I have one 24V line going to it from the power supply and am not using the 24V supply from the fan connector, only the Ground to turn them on and off.

iagobispo, 2023-11-16:
can share your file printer.cfg?

darkraven0386, 2024-12-27:
> can share your file printer.cfg?

X2 :)