makerbase-mks/MKS-SKIPR #83: Where to connect filament runout sensor?

merlinbk, 2024-06-19:
Where to connect Ender 3 filament runout sensor?

merlinbk, 2024-06-20:
It's a 4 Pin Connecter with 3 Pins.

alexvnesta, 2024-07-08:
I'm assuming you figured it out by now - but you will have to re-pin to a 3 pin connector. Generally the 3-pin endstop connectors can be used for filament runout. You can find the location of the endstop sockets on the main page of this git (location 4 in the diagram). Any endstop should work. Check the schematic and make sure that you put V, G, and S in the correct locations. The pinout for that is here: https://github.com/makerbase-mks/MKS-SKIPR/blob/main/hardware/MKS%20SKIPR%20V1.0_002/MKS%20SKIPR%20V1.0_002%20PIN.pdf

This guy has a fantastic video series showing you how to get up and going with the board.

https://www.youtube.com/watch?v=DhbifPQ6ljM